grblHAL/Plugins_spindle #8: Spindle RPM override not taking effect.

andrewmarles, 2022-04-15:
Apologies, I can only test this with my vfd_support branch modifications and my GS20 VFD , but I just wanted to report my experiences with the new spindle switching feature.

https://github.com/Expatria-Technologies/Plugins_spindle

The switching works as expected, both with the M code and with the startup spindle setting.  And when the device is idle or in hold, I can use the spindle speed override and it works as expected.  However, if we are in a run state it seems the spindle override is not taking effect.  I will try to get some better insights into what could be going on.

Thanks as always for these cool new features.

terjeio, 2022-04-16:
Spindle override works for me when in run state. Tested with a STM32F446 Nucleo board and my CNC machine simulator.

andrewmarles, 2022-04-18:
I have continued to chase this down, and it looks to me like what is happening is that the spindle RPM is being polled at a very high rate.  I can see this activity on the modbus port.  Perhaps this is saturating the modbus buffer and preventing the non-blocking updateRPM messages that are used during RUN mode from being transmitted to the VFD?  I have added a timing check on the modbus_send() in the SpindleGetState function that I am using and this does 'fix' the issue in that I can now successfully adjust the spindle speed during a run state.

https://github.com/Expatria-Technologies/Plugins_spindle/blob/ac40ee2937c8e07eb4c85ec5d018723fa0e76ff3/MODVFD.c#L171

But I am not sure this is the best solution to this particular issue?  Is there a better place where we can modulate the rate of spindle RPM requests?

terjeio, 2025-12-23:
> Is there a better place where we can modulate the rate of spindle RPM requests?

See [this comment](https://github.com/grblHAL/Plugins_spindle/issues/42#issuecomment-3682261902).

I am closing this, add to the issue linked above if need be.