grblHAL/Plugins_spindle #10: modbus is a universal interface?

UberChmonya, 2022-05-17:
the name of my VFD is Delta VFD015EL21A. Will it work?

terjeio, 2022-05-17:
Modbus is a communications protcol so yes it is universal. However each VFD vendor implements their own messages on top of that so a separate plugin is needed for each VFD (or each VFD vendor depending on whether they have standardized the messages across models).
Adding a plugin is not too hard if documentation is available. There is a [PR](https://github.com/grblHAL/Plugins_spindle/pull/9) for further VFDs pending, I will soon review that for merging. This may give you an idea of the work involved.

andrewmarles, 2022-05-17:
If your VFD supports true modbus you can very likely get it working with the MODVFD mode from the above PR.

closed 2023-01-25 by terjeio