grblHAL/Plugins_spindle #11: Spindle load feedback with multiple spindles

dresco, 2022-08-29:
Am going to have another look at spindle load reporting for VFD spindles, this time just as a percentage (following discussion in previous [PR](https://github.com/grblHAL/Plugins_spindle/pull/6), and in line with how it's done on other machines).

Am planning to query the maximum amps during spindle configuration, and then the actual amps periodically along with the rpm. Can then use a calculated load percentage for feedback in the realtime report etc. So far so good..

I have a question though, I only want to add this additional report data if the spindle is selected. For instance if I switch away from the Huanyang spindle to the PWM spindle with M104, then I'd want to stop adding the data in the Huanyang plugin realtime report. Is there a way within the spindle plugin to determine whether the spindle is currently active?

terjeio, 2022-08-29:
Hook your code into the [grbl.on_spindle_select](https://github.com/grblHAL/core/blob/5dd9c3eb18b6a0757de65d3fa49912882489355b/core_handlers.h#L125) event?

dresco, 2022-08-29:
Ah yes thanks, so I assume I should just be able to filter on (for instance) the v1_active flag in the realtime report? 

But on checking, something looks off... v1_active is set false when the spindle is Huanyang and true when PWM?

Building with VFD_ENABLE = -1, and N_SPINDLE = 2
```
$395: Default spindle:
    0 - PWM
    1 - Huanyang v1
```

terjeio, 2022-08-30:
Oops - a bug in the [registration code](https://github.com/grblHAL/Plugins_spindle/blob/a3450d236c09818e8fef1c25b9385683993f00a0/vfd/spindle.c#L58-L69), change to:

```
spindle_id_t spindle_register (const spindle_ptrs_t *spindle, const char *name)
{
    if(n_spindle == 0)
        memcpy(&hal.spindle, spindle, sizeof(spindle_ptrs_t));

    if(n_spindle < N_SPINDLE && settings_add_spindle_type(name)) {
        spindles[n_spindle++] = spindle;
        return n_spindle - 1;
    }

    return -1;
}
```

The idea behind having a wrapper for VFD registration was to have a place to add common VFD extension code, e.g. by adding VFD specific function pointers by wrapping the "standard" registration struct in a super struct together with a VFD specific struct. The registration function above would then pass on the "standard" struct to the core and save the VFD specific one to the spindles array. By subscribing to the `on_spindle_select` event in [vfd/spindle.c](https://github.com/grblHAL/Plugins_spindle/blob/master/vfd/spindle.c) it would become easy to add VFD specific functionality? Perhaps this is what you want?

dresco, 2022-08-30:
Thanks, this is what I've got so far.. If you're generally in agreement with the approach, I can open a PR for discussion?

```
diff --git a/vfd/huanyang.c b/vfd/huanyang.c
index 333dd1f..f150f5b 100644
--- a/vfd/huanyang.c
+++ b/vfd/huanyang.c
@@ -39,6 +39,8 @@ static on_report_options_ptr on_report_options;
 static on_spindle_select_ptr on_spindle_select;
 static driver_reset_ptr driver_reset;
 static uint32_t rpm_max = 0;
+static float amps_max = 0;
+static float amps = 0;
 
 static void rx_exception (uint8_t code, void *context);
 static spindle_data_t *spindleGetData (spindle_data_request_t request);
@@ -84,6 +86,25 @@ static void v1_spindleGetMaxRPM (void)
     modbus_send(&cmd, &v1_callbacks, true);
 }
 
+// Read maximum configured current from spindle, value is used later for calculating spindle load
+static void v1_spindleGetMaxAmps (void)
+{
+    modbus_message_t cmd = {
+        .context = (void *)VFD_GetMaxAmps,
+        .adu[0] = vfd_config.modbus_address,
+        .adu[1] = ModBus_ReadCoils,
+        .adu[2] = 0x03,
+        .adu[3] = 0x8E, // PD142
+        .adu[4] = 0x00,
+        .adu[5] = 0x00,
+        .tx_length = 8,
+        .rx_length = 8
+    };
+
+    modbus_set_silence(&v1_silence);
+    modbus_send(&cmd, &v1_callbacks, true);
+}
+
 static void v1_spindleSetRPM (float rpm, bool block)
 {
     if (rpm != rpm_programmed) {
@@ -151,7 +172,7 @@ static spindle_data_t *v1_spindleGetData (spindle_data_request_t request)
 // Returns spindle state in a spindle_state_t variable
 static spindle_state_t v1_spindleGetState (void)
 {
-    modbus_message_t mode_cmd = {
+    modbus_message_t rpm_cmd = {
         .context = (void *)VFD_GetRPM,
         .crc_check = false,
         .adu[0] = vfd_config.modbus_address,
@@ -162,7 +183,19 @@ static spindle_state_t v1_spindleGetState (void)
         .rx_length = 8
     };
 
-    modbus_send(&mode_cmd, &v1_callbacks, false); // TODO: add flag for not raising alarm?
+    modbus_send(&rpm_cmd, &v1_callbacks, false); // TODO: add flag for not raising alarm?
+
+    modbus_message_t amps_cmd = {
+        .context = (void *)VFD_GetAmps,
+        .crc_check = false,
+        .adu[0] = VFD_ADDRESS,
+        .adu[1] = ModBus_ReadInputRegisters,
+        .adu[2] = 0x03,
+        .adu[3] = 0x02,     // Output amps * 10
+        .tx_length = 8,
+        .rx_length = 8
+    };
+    modbus_send(&amps_cmd, &v1_callbacks, false); // TODO: add flag for not raising alarm?
 
[... 101 lines cut ...]
--- a/vfd/spindle.h
+++ b/vfd/spindle.h
@@ -55,7 +55,9 @@ typedef enum {
     VFD_GetMaxRPM,
     VFD_GetMaxRPM50,
     VFD_GetStatus,
-    VFD_SetStatus
+    VFD_SetStatus,
+    VFD_GetMaxAmps,
+    VFD_GetAmps
 } vfd_response_t;
 
 typedef struct {
@@ -75,6 +77,9 @@ typedef struct {
 
 extern vfd_settings_t vfd_config;
 
+extern bool has_spindle_load;
+extern uint8_t spindle_load;
+
 spindle_id_t vfd_register (const spindle_ptrs_t *spindle, const char *name);
 
 #endif

```

terjeio, 2022-08-30:
Here is how I am thinking:

[vfd.zip](https://github.com/grblHAL/Plugins_spindle/files/9451872/vfd.zip)

This way it will be easy to add new (and optional) functionality by letting the VFD drivers telling _vfd/spindle.c_ what they are capable of. As a bonus adding functions to `vfd_ptrs_t` will not break any existing drivers .

dresco, 2022-08-30:
> Here is how I am thinking:

Cool thanks, will have a look...

dresco, 2022-08-30:
This is working for me, just a couple of small updates;

Added the power related modbus bits back for the Huanyang v1 VFD
Additional check that vfd_spindle is not null (else got load report when PWM spindle active)
Add new function to retrieve load value (e.g. from my control panel plugin)

[vfd.zip](https://github.com/grblHAL/Plugins_spindle/files/9454926/vfd.zip)

terjeio, 2022-08-30:
I have commited my changes, can you add yours in a PR?

> Additional check that vfd_spindle is not null (else got load report when PWM spindle active)

I have changed the final code a bit, should be ok now.

> Add new function to retrieve load value (e.g. from my control panel plugin)

Perhaps a function to retrieve the active VFD function pointers (as a pointer to [vfd_spindle](https://github.com/grblHAL/Plugins_spindle/blob/c1095d4be1d77c0c4bef2df94d8cb6ae83063a1c/vfd/spindle.c#L51)) is more flexible? Then the plugin code can determine what is available and any new functions will be instantly available. A bit like how it is done [in the core](https://github.com/grblHAL/core/blob/66f821d8e08996fc290c6d7b2321a2742c54569a/spindle_control.c#L97-L108) but with a single call?