grblHAL/Plugins_spindle #21: VFD as default spindle ($395) does not expose VFD settings

andrewmarles, 2023-07-31:
The dynamic spindle selection seems to work well for VFD spindles when they are selected as spindle > 1.  The is_vfd_spindle and _is_modvfd_selected functions expose the appropriate settings and they appear to work as expected.

But if I set the default spindle to be one of the VFD spindles, it never enumerates correctly.  $476 is never exposed, nor do the MODVFD settings if that spindle is selected.  It does appear that the MODVFD spindle is in fact being activated by the core as I can see traffic on the RS485 interface and it generates error14 when it can't communicate.  I have spent a fair amount of time trying to figure out where it is going wrong but no success yet.

terjeio, 2023-07-31:
> But if I set the default spindle to be one of the VFD spindles, it never enumerates correctly. $476 is never exposed

`$460` is for setting the modbus address if a single VFD spindle is selected (for backwards compatibility). Here is what I get in ioSender when I compile locally (STM32F4xx) with `#define VFD_ENABLE 5`:

[image]

Are you compiling with the Web Builder?

andrewmarles, 2023-07-31:
I am compiling from source (plugins plus UF2 bootloader packaging).  I am using VFD_ENABLE of -1 and N_SPINDLE of 8 so that support for all the different VFDs is compiled in and the user can change in software.

terjeio, 2023-08-01:
Oops, I am messing up a bit, too much going on... Try with this _spindle/select.c_:

[select.zip](https://github.com/grblHAL/Plugins_spindle/files/12224864/select.zip)

I'll be away for a few days so do not have time for much testing.

andrewmarles, 2023-08-01:
This seems to have resolved the issue.  Thanks for the quick fix, I figured it was something simple and I just wasn't seeing it.

terjeio, 2023-08-06:
[Fix commited](https://github.com/grblHAL/Plugins_spindle/commit/379cf2aefc95ff556e5e93ca246f3688c7928e19#diff-20cc0d497fc672ecd03f118eb331e9a7a95331e2fb8a917b72c1f3572a24d57e).

closed 2023-08-06 by terjeio