grblHAL/Plugins_spindle #22: H100 VFD

ennio64, 2023-09-15:
The board is MKS Tinybee.

I have several problems with this driver:
- does not increase speed (always works at the min speed set in the register (my set is 7200 rpm));
- M5 does not always stop the spindle, sometimes it stops with M3 S0 or M3 S0 must precede M5;
- M3 and M4 always rotate in the same direction.

I analyzed the data flow on Modbus by comparing a working configuration to verify what happens (I attach the files).
[H100 whith FluidNC.txt](https://github.com/grblHAL/Plugins_spindle/files/12616597/H100.whith.FluidNC.txt)
[H100 whith grblHAL.txt](https://github.com/grblHAL/Plugins_spindle/files/12616604/H100.whith.grblHAL.txt)
 
Whith grblHAL the commands M3, M4 and M5 are received correctly, but he speed S(xxxx) is written in the register with address 512 instead of 513, and I think it is the cause of the speed limitation and maybe also the other problems.

This log is when IOSENDER send M3 S10000  (firmware fluiNC):

[14/09/2023 15:36:49] 
Modbus Response (COM11) 
Address: 1 
Function: 6 (0x06) - Write Single Register 
    Address: 513 
    Value: 1666 
Checksum: 0x5a73 - OK 
  
[14/09/2023 15:36:50] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 05 c2 06 82  
        Register0: 1474 
        Register1: 1666 
Checksum: 0xd975 - OK 
 
[14/09/2023 15:36:50] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 06 6f 06 82  
        Register0: 1647 
        Register1: 1666 
Checksum: 0x48d0 - OK 
 
[14/09/2023 15:36:50] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 06 82 06 82  
        Register0: 1666 
        Register1: 1666 
Checksum: 0xd8e5 - OK 

**The spindle run at 10000 rpm correctly.**

This log is when IOSENDER send M3 S10000  (firmware grblHAL):

(send M3 S10000)
 
[14/09/2023 17:21:16] 
Modbus Response (COM11) 
Address: 1 
Function: 5 (0x05) - Write Single Coil 
    Address: 73 
    Value: 65280 
Checksum: 0x5dec - OK 
  
[14/09/2023 17:21:16] 
Modbus Response (COM11) 
Address: 1 
Function: 6 (0x06) - Write Single Register 
    Address: 512 <<<<<<<<<<<<<<<<< (not is 513)
    Value: 1670 
Checksum: 0x0a70 - OK 
  
[14/09/2023 17:21:16] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 00 73 00 32  
        Register0: 115 
        Register1: 50 <<<<<<<<<<<<<<<<<<< (always locked to 50)
Checksum: 0x8b8a - OK 
 
[14/09/2023 17:21:17] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 02 af 00 32  
        Register0: 687 
        Register1: 50 
Checksum: 0x4bc8 - OK 
 
[14/09/2023 17:21:18] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 04 b0 00 32  
        Register0: 1200 <<<<<<<<<<<<<<<<<<< (instant speeds 1200=7200 rpm)
        Register1: 50 
Checksum: 0x7a86 - OK 
 
[14/09/2023 17:21:22] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 04 b0 00 32  
        Register0: 1200 
        Register1: 50 
Checksum: 0x7a86 - OK 

**Register1 (which should set the speed to reach) is blocked at 50. The splindle run to min speed 7200 rpm**

I attach the manual of my VFD
[new H100 manual V1.8.pdf](https://github.com/grblHAL/Plugins_spindle/files/12616782/new.H100.manual.V1.8.pdf)

Is there anyone who uses this driver or has done any testing?
Thanks

terjeio, 2023-09-15:
MODVFD could not be used?

ennio64, 2023-09-15:
I tried but I can't make it work. If it works with FluidNC it must also work with grblhal the Vfd H100 is the same.

ennio64, 2023-09-15:
I think that the problem is the 512 address

ennio64, 2023-09-15:
The commands that the grblhal driver sends to the Vfd are the same as those sent by fluidnc, except that it writes the speed to be reached in address 512 instead of 513.

terjeio, 2023-09-15:
Try this binary:

[app-template.zip](https://github.com/grblHAL/Plugins_spindle/files/12619937/app-template.zip)

ennio64, 2023-09-18:
Perfect. Work perfect.

**>>> send M3 S7200**

[18/09/2023 09:17:11] 
Modbus Response (COM11) 
Address: 1 
Function: 5 (0x05) - Write Single Coil 
    Address: 73 
    Value: 65280 
Checksum: 0x5dec - OK 
 
[18/09/2023 09:17:11] 
Modbus Response (COM11) 
Address: 1 
Function: 6 (0x06) - Write Single Register 
    Address: 513 
    Value: 1202 
Checksum: 0x5b07 - OK 

[18/09/2023 09:17:11] 
Modbus Response (COM11) 
Address: 1 
Function: 6 (0x06) - Write Single Register 
    Address: 513 
    Value: 1202 
Checksum: 0x5b07 - OK 
 
[18/09/2023 09:17:11] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 00 00 04 b2  
        Register0: 0 
        Register1: 1202 
Checksum: 0x7931 - OK 

[18/09/2023 09:17:11] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 0 
    Values:  
Checksum: 0x71cb - OK 

[18/09/2023 09:17:11] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 00 db 04 b2  
        Register0: 219 
        Register1: 1202 
Checksum: 0x090a - OK 

[18/09/2023 09:17:12] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 02 59 04 b2  
        Register0: 601 
        Register1: 1202 
Checksum: 0xa89a - OK 
 
[18/09/2023 09:17:12] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 03 d8 04 b2  
        Register0: 984 
        Register1: 1202 
Checksum: 0xf94e - OK 
 
[18/09/2023 09:17:13] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 04 b2 04 b2  
        Register0: 1202 
        Register1: 1202 
Checksum: 0xd826 - OK 

**(The spindle run 7200 rpm)**

**>>> send M3 S10000**

[18/09/2023 09:17:19] 
Modbus Response (COM11) 
Address: 1 
Function: 5 (0x05) - Write Single Coil 
    Address: 73 
    Value: 65280 
Checksum: 0x5dec - OK 

[18/09/2023 09:17:19] 
Modbus Response (COM11) 
Address: 1 
Function: 6 (0x06) - Write Single Register 
    Address: 513 
    Value: 1670 
Checksum: 0x5bb0 - OK 

[18/09/2023 09:17:19] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 04 b2 06 86  
        Register0: 1202 
        Register1: 1670 
Checksum: 0xd891 - OK 

[18/09/2023 09:17:19] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 0 
    Values:  
Checksum: 0x71cb - OK 

[18/09/2023 09:17:19] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 04 c8 06 86  
        Register0: 1224 
        Register1: 1670 
Checksum: 0xf948 - OK 

[18/09/2023 09:17:20] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 05 78 06 86  
        Register0: 1400 
        Register1: 1670 
Checksum: 0xf953 - OK 

[18/09/2023 09:17:20] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 06 86 06 86  
        Register0: 1670 
        Register1: 1670 
Checksum: 0x98e7 - OK 

**(The spindle run 10000 rpm)**

**>>> send M5**

[18/09/2023 09:17:24] 
Modbus Response (COM11) 
Address: 1 
Function: 5 (0x05) - Write Single Coil 
    Address: 75 
    Value: 65280 
Checksum: 0xfc2c - OK 

[18/09/2023 09:17:24] 
Modbus Response (COM11) 
Address: 1 
Function: 6 (0x06) - Write Single Register 
    Address: 513 
    Value: 1200 
Checksum: 0xdac6 - OK 

[18/09/2023 09:17:24] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 06 86 04 b0  
        Register0: 1670 
        Register1: 1200 
Checksum: 0x1991 - OK 

[18/09/2023 09:17:24] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 06 01 04 b0  
        Register0: 1537 
        Register1: 1200 
Checksum: 0xa9b8 - OK 

[18/09/2023 09:17:24] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 05 44 04 b0  
        Register0: 1348 
        Register1: 1200 
Checksum: 0xb829 - OK 

[18/09/2023 09:17:25] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 03 c5 04 b0  
        Register0: 965 
        Register1: 1200 
Checksum: 0xe889 - OK 

[18/09/2023 09:17:25] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 03 06 04 b0  
        Register0: 774 
        Register1: 1200 
Checksum: 0x18b5 - OK 

[18/09/2023 09:17:25] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 02 47 04 b0  
        Register0: 583 
        Register1: 1200 
Checksum: 0x495d - OK 

[18/09/2023 09:17:26] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 00 c8 04 b0  
        Register0: 200 
        Register1: 1200 
Checksum: 0x790e - OK 

[18/09/2023 09:17:26] 
Modbus Response (COM11) 
Address: 1 
Function: 4 (0x04) - Read Input Registers 
    Byte Count: 4 
    Values: 00 00 04 b0  
        Register0: 0 
        Register1: 1200 
Checksum: 0xf8f0 - OK 

**(The spindle is stopped)**

M3, M4 and M5 now work very fine.

Thanks.

closed 2024-08-12 by terjeio