grblHAL/Plugins_spindle #23: $465 and $466

GameOEver, 2024-01-04:
Hello, 

i am trying to get a SIEMENS VFD running but $465 and $466 seem to only accept values up to 255.

In my case the CW and CCW commands would be 1151 and 3199 sice the vfd uses bit 11 for direction.

If i set $465 and $466  to the values 1151 and 3199 it always writes 127 to the modbus register.

Is there any workaround for that?

Thanks in advance!

terjeio, 2024-01-05:
Which version (from $I output)?
I committed a fix 20230130 in my local repo so should be ok in the latest from github.

GameOEver, 2024-01-05:
$1 gives me 25 as output

terjeio, 2024-01-05:
$I not $1...

GameOEver, 2024-01-05:
my bad, was on my way to work. Will check that later.

GameOEver, 2024-01-05:
Hello, 

the firmware of my controller had an older firmware in it. Updating it to the latest beta pre-release solved the issue.

Thank you

closed 2024-01-05 by GameOEver