grblHAL/Plugins_spindle #28: Stepper Spindle Enable signal to be isolated from axis enable

karoria, 2024-04-04:
I have just started testing the stepper spindle. The obvious practical issue I feel is spindle enable and axis enable both are hard bound somewhere in code, so I cant really move an axis to a precise location while rotating the spindle by hand to see if the tool touched the surface. All the other parameters like Spindle enable, direction, rpm, acceleration, works as expected.

karoria, 2024-04-05:
That works. Thanks.

closed 2024-04-05 by karoria

karoria, 2024-04-11:
Hi @terjeio 
After these changes related to stepper spindle, I have noticed my  default pwm spindle and modbus spindle both have issues. FYI, I use only 1 spindle in my machine. I checked above vfds one by one. Above vfds work well on table test but in machine it starts and stops without any notification. I observed message starting with grblHAL and version that time. That should also mean it is resetting when I start spindle. I just doubt on firmware part, it can be another issue also.

terjeio, 2024-04-12:
You have [more than one spindle](https://github.com/grblHAL/core/blob/626ecd06f1181bfbefd8919ea4e3f1698ed5926e/config.h#L61-L66) active at the same time?

karoria, 2024-04-13:
No. I use only one spindle. I have 3 machines now, each running different spindle configuration. PWM, Modbus and Stepper. My point is, after compiled for stepper spindle, when I compile for modbus or pwm spindle, it doesn't work smoothly. Pwm stops itself without any alarm or so. Modbus communication works intermittently in another machine. I think the grblHAL resets after I start spindle. With the same hardware, old builds run perfectly on pwm spindle.

terjeio, 2024-04-13:
All are running on iMXRT1062?

karoria, 2024-04-13:
Yes

terjeio, 2024-04-15:
I have run bech tests for both PWM and VFD (Huanyang v1) spindles and cannot detect any issues.
When PWM stops does the enable signal also change to off? Does it happen when the machine is just sitting idle? Or only when moving?
Which old version works?