grblHAL/Plugins_spindle #31: Stepper Spindle Unexpected Behaviour

engigeer, 2024-08-29:
Testing out the stepper spindle on a FlexiHAL controller board and encountered two unexpected behaviours.

First, ioSender seems to recognize that a 4th-axis is present (and accept G01 A## movement commands). N_AXIS is set to 4 in the config, but I expected that the spindle would “claim” the 4th-axis and so I would only have XYZ movement commands available.

Second (and somewhat related), to move the stepper with a G01 A## command, it seems necessary to invert the stepper enable signal ($4) - which is expected with the DM556 driver I used for this test. However, the stepper will only move with an M03 command if the invert signal enable mask is false (opposite of above).

I understand this driver is experimental, but figured it was worth creating an issue in case others are trying things out and encounter something similar.

terjeio, 2024-09-01:
> First, ioSender seems to recognize that a 4th-axis is present (and accept G01 A## movement commands). N_AXIS is set to 4 in the config, but I expected that the spindle would “claim” the 4th-axis and so I would only have XYZ movement commands available.

A final version will not allow this. It will take a bit of time complete since quite a bit of code has to be added and I have not decided yet how to best implement it. "Claiming" an axis was just a quick and dirty hack to get it working fast.

> Second (and somewhat related), to move the stepper with a G01 A## command, ...

Don't - it is not supported and will not be.

engigeer, 2024-09-01:
Sounds good. 

For what it’s worth, I think it could be potentially useful if it remains possible to also address the stepper spindle as a 4th axis. I’ve got some experience with an industrial lathe that has this feature and found it to be surprisingly useful.

In either case, totally appreciate that finalizing the driver is a non-trivial effort and it will remain experimental for now. Thanks for the insights!

closed 2024-09-01 by engigeer