grblHAL/Plugins_spindle #32: Is Alarm 14 expected following soft reset?

dresco, 2024-10-01:
I'd noticed that I'd always end up with a spindle alarm when stopping a running g-code program in my usual sender (CNCjs). 

Digging a bit further, I see that it's sending a soft-reset when the 'stop' button in the GUI is pressed, and am able to replicate this behaviour by sending Ctrl-X from a terminal console while the spindle is running.

Just wondering whether this is expected behaviour for a (Huanyang) modbus spindle? Thanks!

terjeio, 2024-10-02:
> Just wondering whether this is expected behaviour for a (Huanyang) modbus spindle?

It is not. I do not have any VFD spindles so I test with a simulator - and that responds in a way that does not cause an alarm. Do you get an alarm when booting the controller as well?

If you comment out one/both of [these lines](https://github.com/grblHAL/Plugins_spindle/blob/e308e30a119afc0c29fa0a4993fb541019f41b1b/vfd/huanyang.c#L283-L284) does it still alarms?

dresco, 2024-10-02:
> Do you get an alarm when booting the controller as well?

Not when the VFD is already powered on & ready.

> It is not.

Thanks for the confirmation, I have a spare Huanyang VFD with my dev setup, so will wait for a rainy day and debug there..

dresco, 2024-10-02:
Couldn't help myself, and had a quick look when I should be doing something more practical in the nice weather ;)

Yes, if I comment out those lines then it doesn't error.. I believe the problem is [here](https://github.com/grblHAL/Plugins_spindle/blob/e308e30a119afc0c29fa0a4993fb541019f41b1b/modbus_rtu.c#L343-L364) in modbus_reset(). 

What I think is happening is that the spindle is commanded to stop (spindleSetState / spindleSetRPM), then modbus_reset() immediately forces the state back to Modbus_Idle, then onDriverReset() tries to query the limits etc. This is ignored by the VFD - as it's still within the idle timeout from the previous spindleSetRPM, triggering the alarm.

See attached capture from the RS485 pair, around the 350-400ms timeframe. I believe these are the VFD_SetStatus command & response, the VFD_SetRPM command & response, and the VFD_GetRPMAt50Hz command immediately afterwards;

[image]

I think an easy fix would be to move the `while(state != ModBus_Idle)` loop above the `if(sys.abort)` check, unless you think otherwise? Btw, I'm always amazed it works at all when I look at the common mode noise from these cheap VFDs!

dresco, 2025-01-05:
Fixed by c3020ab, thanks!

closed 2025-01-05 by dresco