grblHAL/Plugins_spindle #35: All spindles share the same settings - why?

axg20202, 2024-11-20:
I have 2 spindles, one PWM and the other is via ModBus VFD. For some reason, there is a single set of parameters (speed, frequency etc) which affects all spindles. This doesn't seem right. For example, my PWM spindle needs a speed settings of 0-1000, while my Modbus spindle needs 1200 to 24000 rpm.  There really needs to be dedicated settings for each spindle.

terjeio, 2024-11-20:
> the other is via ModBus VFD.

Which?

> There really needs to be dedicated settings for each spindle.

Nope, RPM range should be read from the VFD, [like this](https://github.com/grblHAL/Plugins_spindle/blob/23b415712b9cff700ee9dd6978888189f3429fcd/vfd/h100.c#L51-L73), and the VFD drivers that currently do not should be updated.
If settings were to be added for VFD drivers then the setting values has to be transferred to the VFD, and currently I do not have any to test against.

axg20202, 2024-11-20:
Ah, OK. So VFD/modbus spindles don't use the rpm setting in grblhal, only the PWM spindle? I could have sworn that changing those parameters affects the speed of my vfd spindle as well. My vfd is HY type connected over modbus. My other spindle is a laser that I believe requires 0 to 1000 rpm setting.

terjeio, 2024-11-21:
Which version are you running (build date in the $i output)?
If a relatively new version `$spindlesh` can be used to output details about the spindles, min/max RPM is output as the two last values for active spindles.
For VFDs the reported RPM range is taken from $30 and $31 initially, and changed to the actual values after reading min/max from the driver.