grblHAL/Plugins_spindle #38: Stepper spindle lathe vision/idea

jollydodo, 2025-05-08:
Hey, 

Perhaps not directly related, but I was wondering what the vision / real world use case with the stepper spindle is. I guess some people already have some experience, but I'm in the process of converting my lathe to a cnc lathe. This lathe (simple Emco Compact) is going to be mainly used for small intricate things and not the heavy stuff. So mainly pom/brass/aluminium. Using a stepper as spindle would simplify things greatly (no need for an encoder) and makes everything more lightweight, but if look at the torque graph of a pretty beefy [closed loop nema34](https://www.omc-stepperonline.com/nl/s-serie-nema-34-gesloten-lus-stappenmotor-12-00nm-1699-34oz-in-encoder-1000ppr-4000cpr-34hs59-6004d-e1k-1m5) then it seems that at 1000rpm there is little  torque (1.5nm) left. So basically my question in what types of setups/lathes would one use a stepper spindle? Is it only for low rpm torque where you have a separate spindle and use the stepper only for threading?  Any input is very much appreciated! I'm quite new with lathe's.

terjeio, 2025-05-16:
>  So basically my question in what types of setups/lathes would one use a stepper spindle?

I have not tried it in my lathe (a converted Emco Compact 5), I use a washing machine motor and a VFD. My guess is that a stepper spindle could be useful in a mini lathe if the torque is enough for the material that needs to be cut. Or it could be somehow connected to the spindle in addition to the main motor so it could be used for indexing? Anyway, I added it because I could - the main part of the code is shared with step injection required for Plasma THC.

paukstelis, 2025-07-25:
I am planning on using stepper spindle as part of a "virtual" Rose Engine. In this case, RPMs stay low with minimal cutting forces, but coordinated movement between the spindle and cartesian axes are required.